fsmMLK/inkscapeCircuitSymbols #2: Crashes on Ubuntu with 0.48.4

mbrunnen, 2017-08-16:
I use the inkscape package 0.48.4-3ubuntu2 and have it installed with inkscapeMadeEasy in `~/.config/inkscape/extensions`:
```
├── extensions
│   ├── circuitSymbols_general.inx
│   ├── circuitSymbols.py
│   ├── circuitSymbols_semiconductors.inx
│   ├── inkscapeMadeEasy_Base.py
│   ├── inkscapeMadeEasy_Draw.py
│   ├── inkscapeMadeEasy_Plot.py
│   └── textextLib
│       ├── basicLatexPackages.tex
│       ├── CircuitSymbolsLatexPreamble.tex
│       ├── __init__.py
│       ├── textext.inx
│       └── textext.py
├── icons
├── keys
├── palettes
├── preferences.xml
└── templates
```
inkscape crashes immediately and returns:
```
Emergency save activated!
Emergency save completed. Inkscape will close now.
If you can reproduce this crash, please file a bug at www.inkscape.org
with a detailed description of the steps leading to the crash, so we can fix it.
zsh: segmentation fault (core dumped)  inkscape
```

mbrunnen, 2017-08-18:
the problematic code is in the `.inx` files:
```xml
    <page name="Colors" appearance="minimal" _gui-text="Colors">
      <param name="desc1" type="description" appearance="header">Current annotation color</param>
      <param name="currColor" type="optiongroup" appearance="minimal" _gui-text="Color:">
        <_option value="#FF0000">default current color</_option>
        <_option value="picker">use color picker</_option>
      </param>
      <param name="colorPickerCurrent" type="color"/>
      <param name="desc1" type="description" appearance="header">Voltage annotation color</param>
      <param name="voltColor" type="optiongroup" appearance="minimal" _gui-text="Color:">
        <_option value="#217B21">default voltage color</_option>
        <_option value="picker">use color picker</_option>
      </param>
      <param name="colorPickerVolt" type="color"/>
    </page>
```
Outcommenting these lines "fixes" the problem.

fsmMLK, 2017-08-18:
It might be an issue with the color picker widget on inkscape v0.48

Try to comment only the followin lines and check whether the extension works or not.
      <param name="colorPickerCurrent" type="color"/>
      <param name="colorPickerVolt" type="color"/>

closed 2018-09-28 by fsmMLK