fsmMLK/inkscapeCircuitSymbols #4: Does not work in ubuntu 17.04

bahram-houchmandzadeh, 2017-11-14:
I installed various module of fsmMLK (MadeEasy, CircuitSymbols, plot functions, ... following exactly the instructions. All the fsmMLK extensions produce the same error (see below), without displaying anything. The standard python in /usr/bin is 2.7 : can that be a problem ? Do these modules use python 3 ?
System : Ubuntu 17.04
Inkscape : 0.92

I have other extensions installed (such as textext), which work fine. 

Here the error message:
=========================
Traceback (most recent call last):
  File "circuitSymbols.py", line 1956, in <module>
    circuit.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
    self.effect()
  File "circuitSymbols.py", line 251, in effect
    currName=so.bipoleRLCCurrName,invertArrows=so.bipoleRLCVoltCurrInvert)
  File "circuitSymbols.py", line 575, in drawCapacitor
    inkDraw.text.latex(self,group,value,pos_text,fontSize=self.fontSize,refPoint='bc',preambleFile=self.preambleFile)
  File "/home/bahram/.config/inkscape/extensions/inkscapeMadeEasy_Draw.py", line 1259, in latex
    BboxMin, BboxMax = ExtensionBaseObj.getBoundingBox(groupLatex)
  File "/home/bahram/.config/inkscape/extensions/inkscapeMadeEasy_Base.py", line 575, in getBoundingBox
    coords = self.getPoints(element)
  File "/home/bahram/.config/inkscape/extensions/inkscapeMadeEasy_Base.py", line 541, in getPoints
    listPoints = self.getPoints(obj)
  File "/home/bahram/.config/inkscape/extensions/inkscapeMadeEasy_Base.py", line 541, in getPoints
    listPoints = self.getPoints(obj)
  File "/home/bahram/.config/inkscape/extensions/inkscapeMadeEasy_Base.py", line 548, in getPoints
    coordsNP = np.hstack((np.array(listCoords), np.ones([len(listCoords), 1]))).transpose()
  File "/usr/lib/python2.7/dist-packages/numpy/core/shape_base.py", line 286, in hstack
    return _nx.concatenate(arrs, 0)
ValueError: all the input arrays must have same number of dimensions

fsmMLK, 2017-12-19:
No they all use python 2.7

Try to disble latex suport and see if the problems disappears.

fsmMLK, 2018-02-04:
Please install package pstoedit and see if it fixes the problem.

alessandroambler, 2018-03-12:
I had the same problem and this fixed it for me. Thanks!

closed 2018-09-28 by fsmMLK