fsmMLK/inkscapeCircuitSymbols #5: Inkscape 0.92 -- 64bit on W7 Enterprise

MikeAnbplips, 2017-11-22:
I am getting the following error in Python code:

```
Traceback (most recent call last):
  File "circuitSymbols.py", line 5, in <module>
    import inkscapeMadeEasy_Base as inkBase
  File "C:\Users\me\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy_Base.py", line 81
    """
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 142-143: truncate
```

Nothing gets drawn.

MikeAnbplips, 2017-11-22:
I've come around this, it is related to triple quoted sections where a \u string is present
(when describing \usepackage). If you escape the slash, the error disappears.

closed 2017-11-22 by MikeAnbplips

fsmMLK, 2017-12-19:
thanks for the report and solution. =)

I also found a few other places where escaping slashes were needed. I suggest you to get the newest verstion on github