fsmMLK/inkscapeCircuitSymbols #8: Inkscape 0.91 on Mac OS X - Can't draw any symbol

cancelliere, 2018-12-16:
I can't get the extension to work.

The error I receive is:
`RuntimeError: Command pdflatex /var/folders/72/msgvrvh51jlbctvzxd9qr8840000gn/T/tmprSk2yD/tmp.tex -interaction=nonstopmode -halt-on-error failed: [Errno 2] No such file or directory`

If I comment the `remove_temp_files(self)` method in `textext.py` (lines 515 to 518) and run the `pdflatex <tempfile.tex> <switches>` command in a terminal, I can compile the .tex file correctly.

If I disable Latex altogether, I can get the symbol with the raw string as a value.

I'm using Inkscape 0.91+devel+osxmenu r12922 on Mac OS X 10.11.6.

Full traceback here:
`Traceback (most recent call last):
  File "circuitSymbols.py", line 2264, in <module>
    circuit.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 269, in affect
    self.effect()
  File "circuitSymbols.py", line 247, in effect
    currName=so.bipoleRLCCurrName,invertArrows=so.bipoleRLCVoltCurrInvert)
  File "circuitSymbols.py", line 568, in drawResistor
    inkDraw.text.latex(self,group,value,pos_text,fontSize=self.fontSize,refPoint='bc',preambleFile=self.preambleFile)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkscapeMadeEasy_Draw.py", line 1230, in latex
    tex.affect([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath], output=False)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 269, in affect
    self.effect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/textextLib/textext.py", line 151, in effect
    self.options.scale_factor, converter_cls, old_node)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/textextLib/textext.py", line 165, in do_convert
    new_node = converter.convert(text, preamble_file, scale_factor)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/textextLib/textext.py", line 532, in convert
    self.tex_to_pdf(latex_text, preamble_file)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/textextLib/textext.py", line 509, in tex_to_pdf
    exec_command(['pdflatex', self.tmp('tex')] + latexOpts)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/textextLib/textext.py", line 370, in exec_command
    raise RuntimeError("Command %s failed: %s" % (' '.join(cmd), e))
RuntimeError: Command pdflatex /var/folders/72/msgvrvh51jlbctvzxd9qr8840000gn/T/tmprSk2yD/tmp.tex -interaction=nonstopmode -halt-on-error failed: [Errno 2] No such file or directory`

fsmMLK, 2018-12-26:
Hum... I wonder whether Mac OS has problems with python's 'tempfile' module

vitorsmonteiro, 2019-03-15:
I'm having a similar issue in Inkscape0.92/Windows10 when using the general circuit extension. Here is the error message;

> Traceback (most recent call last):
>   File "circuitSymbols.py", line 2264, in <module>
>     circuit.affect()
>   File "inkex.py", line 283, in affect
>     self.effect()
>   File "circuitSymbols.py", line 247, in effect
>     currName=so.bipoleRLCCurrName,invertArrows=so.bipoleRLCVoltCurrInvert)
>   File "circuitSymbols.py", line 568, in drawResistor
>     inkDraw.text.latex(self,group,value,pos_text,fontSize=self.fontSize,refPoint='bc',preambleFile=self.preambleFile)
>   File "inkscapeMadeEasy_Draw.py", line 1230, in latex
>     tex.affect([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath], output=False)
>   File "inkex.py", line 283, in affect
>     self.effect()
>   File "textextLib/textext.py", line 126, in effect
>     % ';\n'.join(converter_errors))
> RuntimeError: No Latex -> SVG converter available:
> PstoeditPlotSvg: Command pstoedit -help failed: [Error 2] The system cannot find the file specified;
> SkConvert: Command pstoedit failed: [Error 2] The system cannot find the file specified;
> Pdf2Svg: Command pdf2svg failed: [Error 2] The system cannot find the file specified

When using the semiconductor it works fine as long as I uncheck the "Add voltage arrow" and "Add voltage box". When they are selected a error appears. Here is the complete error message from when those checkbox are selected:

> Traceback (most recent call last):
>   File "circuitSymbols.py", line 2264, in <module>
>     circuit.affect()
>   File "inkex.py", line 283, in affect
>     self.effect()
>   File "circuitSymbols.py", line 390, in effect
>     currName=so.diodeCurrName,invertArrows=not so.diodeVoltCurrInvert,flagType=so.diode,mirror=so.diodeMirror)  
>   File "circuitSymbols.py", line 832, in drawDiode
>     self.drawVoltArrow(group,[position[0]+25 ,position[1]+7],name=voltName,color=self.voltageColor,angleDeg=angleDeg,invertArrows= not (invertArrows != mirror ))
>   File "circuitSymbols.py", line 2166, in drawVoltArrow
>     refPoint=justif,textColor=color,angleDeg=-theta,preambleFile=self.preambleFile)
>   File "inkscapeMadeEasy_Draw.py", line 1230, in latex
>     tex.affect([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath], output=False)
>   File "inkex.py", line 283, in affect
>     self.effect()
>   File "textextLib/textext.py", line 126, in effect
>     % ';\n'.join(converter_errors))
> RuntimeError: No Latex -> SVG converter available:
> PstoeditPlotSvg: Command pstoedit -help failed: [Error 2] The system cannot find the file specified;
> SkConvert: Command pstoedit failed: [Error 2] The system cannot find the file specified;
> Pdf2Svg: Command pdf2svg failed: [Error 2] The system cannot find the file specified

I'm really new at this, so maybe is a simple mistake. I really hope this works, it seems a great tool.
Regards,

fsmMLK, 2019-05-05:
I guess you are having problems with latex support. Did you try to disable Latex support?

follow the instructions here:
https://fsmmlk.github.io/inkscapeMadeEasy/

fsmMLK, 2020-01-05:
I am uploading soon a new version of inksapeMadeEasy that might solve the possible problem with tempfile.

closed 2021-05-01 by fsmMLK