fsmMLK/inkscapeCircuitSymbols #10: Nothing happens when inserting symbol

david-hakansson, 2019-09-17:
Hi,
On Windows 10 Pro, 64-bit system. Tried to install Inkscape 0.92.4, 32 bit version to use the CircuitSymbols. When trying to insert any symbol in InkScape I see a window flashing by really quickly and then disappears. If I press the Apply button repeatedly I'm able to read the text: 'General' working, please wait.

But nothing happens. No symbol is inserted. I have tried with and without LaTex, doesn't make any difference.

Am I doing something wrong?

Best regards.
David

fsmMLK, 2019-12-13:
no idea. I installed it a couple of times on differente Windows machines and they worked fine (no latex)

gijsbent, 2020-03-10:
I experience the same issue, is this problem already solved?

closed 2022-08-07 by fsmMLK