fsmMLK/inkscapeCircuitSymbols #11: Error on inserting symbol: all the input arrays must have same number of dimensions

SapiensAnatis, 2019-11-05:
Full error trace:

```python
Traceback (most recent call last):
  File "circuitSymbols.py", line 2264, in <module>
    circuit.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
    self.effect()
  File "circuitSymbols.py", line 257, in effect
    currName=so.bipoleRLCCurrName,invertArrows=so.bipoleRLCVoltCurrInvert)
  File "circuitSymbols.py", line 674, in drawCapacitor
    inkDraw.text.latex(self,group,value,pos_text,fontSize=self.fontSize,refPoint='bc',preambleFile=self.preambleFile)
  File "/usr/share/inkscape/extensions/inkscapeMadeEasy_Draw.py", line 1259, in latex
    BboxMin, BboxMax = ExtensionBaseObj.getBoundingBox(groupLatex)
  File "/usr/share/inkscape/extensions/inkscapeMadeEasy_Base.py", line 860, in getBoundingBox
    coords = self.getPoints(element)
  File "/usr/share/inkscape/extensions/inkscapeMadeEasy_Base.py", line 826, in getPoints
    listPoints = self.getPoints(obj)
  File "/usr/share/inkscape/extensions/inkscapeMadeEasy_Base.py", line 826, in getPoints
    listPoints = self.getPoints(obj)
  File "/usr/share/inkscape/extensions/inkscapeMadeEasy_Base.py", line 833, in getPoints
    coordsNP = np.hstack((np.array(listCoords), np.ones([len(listCoords), 1]))).transpose()
  File "/usr/lib/python2.7/site-packages/numpy/core/shape_base.py", line 338, in hstack
    return _nx.concatenate(arrs, 0)
ValueError: all the input arrays must have same number of dimensions
```

The extension works correctly if I disable LaTeX support, but I'm creating a diagram for use in a LaTeX document so I'd rather keep it on if at all possible. That suggests that the problem is to do with LaTeX, though.

As per issue #4, I tried installing pstoedit (I didn't have it before), but I still get the same error. Any ideas?

fsmMLK, 2019-12-13:
it seems latex is not creating the pdf of yout latex code;

fsmMLK, 2020-01-05:
I am uploading soon a newer version of InkscapeMadeEasy that might solve this issue.

closed 2020-10-12 by fsmMLK