fsmMLK/inkscapeCircuitSymbols #13: Circuit symbols

LookieGit, 2020-01-07:
Hello,
I'm using Inkscape 92.4 (dutch version) on Win10 and installed the extension "Circuit Symbols" (and InkscapeMadeEasy) exactly like indicated in the readme file and as on your website.
I can see the topic in the menu under "fsmMLK" but I got an error by using it. The error = "Traceback (most recent call last):
  File "circuitSymbols.py", line 8, in <module>
    from drawAmpOp import ampOp
ImportError: No module named drawAmpOp"
I can't find the problem. Does anyone have a solution for that?
Thanks

fsmMLK, 2020-01-07:
You also need to copy the draw*.py files to the extension folder. My instructions were outdated. Please see if this solves the issue.

LookieGit, 2020-01-07:
I tried again, but then I got another message:
Traceback (most recent call last):
  File "circuitSymbols.py", line 577, in <module>
    circuit.affect()
  File "C:\Program Files\Inkscape\share\extensions/inkex.py", line 283, in affect
    self.effect()
  File "circuitSymbols.py", line 427, in effect
    convention=so.switchConvention)
  File "drawSwitches.py", line 104, in drawNPST
    preambleFile=self.preambleFile)
  File "inkscapeMadeEasy_Draw.py", line 1243, in latex
    output=False)
  File "C:\Program Files\Inkscape\share\extensions/inkex.py", line 283, in affect
    self.effect()
  File "textextLib/textext.py", line 125, in effect
    % ';\n'.join(converter_errors))
RuntimeError: No Latex -> SVG converter available:
PstoeditPlotSvg: Command pstoedit -help failed: [Error 2] The system can't find the specified file;
SkConvert: Command pstoedit failed: [Error 2] The system can't find the specified file;
Pdf2Svg: Command pdf2svg failed: [Error 2] The system can't find the specified file

I suppose there is more to do, to make it work.
Thanks for an update.

fsmMLK, 2020-01-11:
Did you install a pstoedit + ghosview? Check if you can run the extension correctly if you disable latex support.

LookieGit, 2020-01-13:
No, I didn't install a pstoedit and Ghosview (I think you mean Ghostview). I only installed the required and indicated things: inkscapeMadeEasy-master and GPL Ghostscript.
It looks that Latex was not properly disabled.
Now it seems to work properly. Item can be closed
Thanks for your help.

closed 2020-01-13 by fsmMLK