fsmMLK/inkscapeCircuitSymbols #14: BJT arrow disappear

its-fonsy, 2020-01-08:
When I have to draw two BJT, after clicking 'Apply' to draw the 2nd one, the arrow on the first one disappear.

fsmMLK, 2020-01-13:
can you attach one svg file with the two BJTs?

its-fonsy, 2020-01-14:
The file is there [ink-bug.tar.gz](https://github.com/fsmMLK/inkscapeCircuitSymbols/files/4058235/ink-bug.tar.gz).
I have also recorded a short GIF to show you what I am doing.
[image]
As you can see when I create the 2nd npn, the first one lose the arrow of the emitter.

fsmMLK, 2020-01-16:
The file you sent me opens fine here on my machine.
Try to restart inkscape and open the file you sent me. See if the arrows appear.

It might be an issue with your inkscape. Which version are you using?

its-fonsy, 2020-01-16:
I am using `Inkscape 0.92.4`, but even with the `0.91` I had the same issue.
I've tried to export the image in `.png` and `.svg` format and both BJTs has arrows. I didn't do it before because I was assuming that Inkscape would have export what I was seeing.

closed 2020-11-10 by fsmMLK