fsmMLK/inkscapeCircuitSymbols #18: latex use is broken with textext 1.3.0 and miktex

PinkMushroom, 2021-02-02:
After a considerable amount of effort making sure textext was working correctly, I still get an indecipherable error when latex is enabled. The good news is that circuitsymbols works if i disable latex.
I would not be surprised if this is the result of yet another missing package in the latex installation (that was the problem i encountered with textext).

```
Traceback (most recent call last):
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\textext\base.py", line 558, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\textext\utility.py", line 240, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command C:\Users\xxxxxx\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "circuitSymbols.py", line 553, in <module>
    circuit.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "circuitSymbols.py", line 272, in effect
    self.drawResistor(root_layer, position, value=so.bipoleRLCVal, angleDeg=so.bipoleRLCRot, flagVolt=so.bipoleRLCVolt,
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\circuitSymbols\drawRLC.py", line 95, in drawResistor
    inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint='bc', preambleFile=self.preambleFile)
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1186, in latex
    tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\textext\base.py", line 315, in effect
    self.do_convert(new_text,
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\textext\base.py", line 388, in do_convert
    converter.tex_to_pdf(tex_executable, text, preamble_file)
  File "C:\Users\xxxxxx\AppData\Roaming\inkscape\extensions\textext\base.py", line 562, in tex_to_pdf
    raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
textext.errors.TexTextConversionError: ! LaTeX Error: File `siunitx.sty' not found.

Type X to quit or <RETURN> to proceed,
```

fsmMLK, 2021-02-02:
siunitx.sty not found. Install siunitx latex package.

PinkMushroom, 2021-02-02:
and there it is at the very end, i just wasn't patient enough to read through all the messages.
You also need the following packages

steinmetz
pict2e

and then success! Thanks for your help.
The latex output really does look good of course, but it's a bit of work to get there...

closed 2021-02-02 by fsmMLK