fsmMLK/inkscapeCircuitSymbols #20: How to troubleshoot

jsavage, 2021-02-23:
Hi, Have just got Circuit Symbols partially working for the first time on Windows 10.
Originally attempted to use 1.02 x64 MSI install of Inkscape. Could not find the extn in the menu. 
Saw reported x64 issue reported by someone else so reverted to Inkscape 1.0 x86 MSI install on a 64 bit Windows 10
Can now see extn listed General and Semiconductor
Selection extension seemed to hang after selecting a diode - fatal.
Restart Inkscape. Had success selecting a resistor.
Attempted to insert an Inductor - extension hung again - fatal.

Grateful for any tips on testing the extension more systematically.

jsavage, 2021-02-23:
Actually, with a lot of patience (waiting) this extension is drawing capacitors, inductors and diodes symbols.
The issue is that a blank interstitial dialog box appears and that gives the appearance of inkscape hanging. 
Just leaving it be after pressing apply and then pressing close after a brief message appears and disappears again results in the selected symbol being drawn in the centre of the diagram  - Not where l last clicked.   On my Core i5 Dell laptop each symbol takes about 35 secs to render.

Do others see sluggish performance or is unusual?  

Screenshot follows:
 
[image]

fsmMLK, 2021-11-16:
it shouldn't take that long to create symbols. I wonder if there is an issue with your installation.
The extension will create the symbol in the center of the screen.

closed 2024-12-25 by fsmMLK