fsmMLK/inkscapeCircuitSymbols #23: Electric symbol requests

fsmMLK, 2021-03-20:
Please leave a message below with electric symbols you would like to have.

sriniv1422, 2021-04-30:
Can you please add these symbols to your circuit library...
[image]

fsmMLK, 2021-04-30:
Oh no! The hidden substrate notation! =S

I will take a look  =)

fsmMLK, 2021-05-01:
sriniv1422, symbols added.

sriniv1422, 2021-05-01:
> sriniv1422, symbols added.
The NMOS is fine , but the PMOS does not turn out to be correct. Actually i would like the style of the BJT to be maintained.
and want the NMOS and PMOS to look like them with the gate line at the centre
[image]

sriniv1422, 2021-05-01:
if u can create a separate category called simple NMOS and Simple PMOS that will be even better

fsmMLK, 2021-05-02:
I am following these two guidelines:

https://en.wikipedia.org/wiki/MOSFET#Circuit_symbols
https://www.d.umn.edu/~sburns/EE2212Fall2012/FETSymbolsTextAndMore.pdf

What is exactly wrong with PMOS?

sriniv1422, 2021-05-02:
fsmMLK

when i use the latest menu this is what i get close to what i want

[image]

notice that the gates of NMOS and PMOS and either low or high and not centered. This is a very old convetion and no one uses these nowadays. 

Actually in the link  https://en.wikipedia.org/wiki/MOSFET#Circuit_symbols
 the NMOS , PMOS gates are at the centre and this is the most commonly used in diagrams , IEEE papers , thesis etc.

the BJT transistors you created are of excellent quality, they turn out to be really good in presentations and papers,
It would be great if you use the same line thickness and arrow size as you used in the BJT , with the gate centered for NMOS /PMOS.
Also , by defualt S is on TOP of Drain for PMOS. this is the most widely used convention. I can get to this by fliiping ,S and D , but the gate is lifted high. 

I really appreciate if you can make them like BJT. it would look awesome

fsmMLK, 2021-05-02:
How does it look now? I made the gate centered for all 3-terminal (no bulk). I am thinking of keeping S/D positions as they are now and the user check to swap S/D if they want. It will make my code cleaner and I don't think this will cause too much trouble to the user.

[image]

Now the linewidth I am not sure. I think uniform line width makes the symbol very 'fat', specially if we draw the body diode or the envelope (see below). Arrow size is slightly smaller in MOS otherwise the arrow would be too big for the tight space of the symbol ( I don't want to make the entire symbol larger just to acommodate the larger arrow.

[image]

sriniv1422, 2021-05-02:
Looks good fsmmlk. please go ahead and make the code change to git hub. Lets keep the thick line . I like it that way better and is consistent . I wont be using the envolope option or body diode anyways. I like to keep it consistent with BJT and for that the thick is perfect. Iam ok with sawpping D/S to get to what I want for PMOS  if that means less code change for you. 
let me know when you are done uploading to github and I can try it out

fsmMLK, 2021-05-02:
Uploaded a new version.

I made the FET use finer lines (0.7) when the envelop is present, otherwise thicker lines (1.0). If you want to have thick lines even with the envelopes, you can edit drawTransistors.py and change  lineWidth to 1.0  (line ~176 )

sriniv1422, 2021-05-02:
fsmmlk ,

Looks good. much appreciate the help...
I wanted to ask you one more thing,

is it possible to create timing diagrams with your package.?
there is nothing out there that can create good timing diagrams except for
wavedrom.
we can talk further in case you are interested to implement it. that would
be a game changer for the entire community.

LEt me know..
with the mods you did , i can now draw good schematics.
thanks a lot..

fsmMLK, 2021-05-02:
Glad to hear that this extension is useful to people.

Now, on timing diagrams, yes, it is possible to create one. I have some questions on that:
1-) What is the problem with wavedrom?
2-) Why would someone need something different? Wavedrom looks quite nice and can generate pretty diagrams.

sriniv1422, 2021-05-02:
fsmmlk

well just wanting to have everything under inkscape.
I agree with you that wavedrom can generate pretty diagrams..
If it is too much work, then no need.

fsmMLK, 2021-05-02:
I found a python module for wavedrom (https://pypi.org/project/wavedrom/). This is the best scenario since there is no need to reinvent the wheel and would make easier to integrate with inkscape.

One idea could be to have an inkscape GUI/text interface for wavedrom embeded in inkscape.

sriniv1422, 2021-05-02:
fsmmlk,

that would be awesome . I did not know that there was a python module
available for wavedrom. If you can integrate it in inkscape , that would be
great. Many people will start using it as everything is now available under
inkscape..

Can you do this ? If so , it will be of immense help.

fsmMLK, 2021-05-02:
I will take a look and create a small proof of concept example. I cannot promisse it will be quick.

I would like to hear from you later what a good interface and features would be nice to have (I don't have experiente with wavedrom)

Perhaps I will create a new private repository only for that and later integrate with circuitsymbols.

sriniv1422, 2021-05-02:
fsmmlk

Sure. In fact it need not be as complex as wavedrom. It is enough if we are
able to create waveforms easily..
I can help you with features and interfaces . Let me know when you have
some thing ready and i can take a look at it.
Take your time, there is no hurry for this.

m3nsh, 2021-06-18:
Can you add RFIC symbols like these, please? https://github.com/cfriesicke/inkscape-schematic-symbols :)

sriniv1422, 2021-06-18:
fsmMLK

have you made any attempts at wavedrom?
Srini

fsmMLK, 2021-07-13:
I have a basic version of a wavedrom extension. The extension requires the user to inform the full string that defines the diagram.

[image]

jm-sanz, 2022-01-21:
Good morning, I wanted to thank you for your work, it is great. I was wondering if you could add some components to your list as this extension is very handy. I attach a .svg file I obtained from a [YouTube video
](https://www.youtube.com/watch?v=AsS4nxFWKQY)

I commonly use magnetics and, by now, this tool only includes one type of coil and it would be great to have more variety (same with other components as power sources as dependent power voltage sources):

[image]

Thank you for your awesome contribution.

[ELECTRICAL SYMBOLS.zip](https://github.com/fsmMLK/inkscapeCircuitSymbols/files/7912246/ELECTRICAL.SYMBOLS.zip)

fsmMLK, 2022-01-23:
Sure, I'm working on it now! =D

what do you mean by "same with other components as power sources as dependent power voltage sources"

jm-sanz, 2022-01-23:
Awesome, thank you very much!.

Sorry I misswrote. I meant that some component, probably not based on official standards have different possibilites for being draw. Examples:

[image]

And would be great to have variable DC power supplies added as in previous comment.

Do you know some way of autojoin the component nets in inkspace?

Thank you :).

fsmMLK, 2022-01-24:
what transformer features would you like to have in this extension? I don't work much with transformers and don't know the features that are worth implementing. At the moment I have implemented:

- iron, air and ferrite core
- option to add 1 central tap per winding
- phase shift indicator
- N1:N2 ration text

jm-sanz, 2022-01-25:
Yes, that would be cool. Maybe some adjustable coils, coils with polarity or transformers with two or more secondary coils (with different cores).

Thank you!.

fsmMLK, 2022-01-27:
This is what I have so far

[image]

jm-sanz, 2022-02-04:
That looks great!, thank you.

TomasVCZ, 2022-02-09:
> I have a basic version of a wavedrom extension. The extension requires the user to inform the full string that defines the diagram.
> 
> [image]

Hello, do you have any update on wavedrom? Looks interesting and I would like to test it.

Thanks, Tomas

sriniv1422, 2022-02-09:
> > I have a basic version of a wavedrom extension. The extension requires the user to inform the full string that defines the diagram.
> > [image]
> 
> Hello, do you have any update on wavedrom? Looks interesting and I would like to test it.
> 
> Thanks, Tomas

Iam also waiting for this . Icannot seem to  find this on got hub.

fsmmlk

is this ready so we can try it . 
Besides , i want to add some more features to able to add analog waveforms as well to this . 
Let us know on how we can try this out on inkscape ?
Srini

fsmMLK, 2022-02-19:
Transformers and variable components are now available! =D

fsmMLK, 2022-02-19:
> > I have a basic version of a wavedrom extension. The extension requires the user to inform the full string that defines the diagram.
> > [image]
> 
> Hello, do you have any update on wavedrom? Looks interesting and I would like to test it.
> 
> Thanks, Tomas

I am returning to it now! It is working as presented in the image. you have to type in the code and the extension creates the diagram. At the moment there is neither a GUI to help creating the text nor a way to verify the input string.

fsmMLK, 2022-02-19:
current wavedrom UNSTABLE version can be found here. Please give me feedback.

 https://github.com/fsmMLK/wavedromScape

josejsarmento, 2022-09-21:
Would it be possible to choose the orientation of the component values? If I rotate 90º a resistor, I would like its name to stay horizontal, for example.