fsmMLK/inkscapeCircuitSymbols #24: scaling of symbols and fonts ?

PinkMushroom, 2021-04-01:
Hi,

I've noticed that the symbols generated are somewhat large-ish.  When i scale them down the text scales down and then becomes too small to read.

I really did read through the documentation for some sort of scaling setting and did not see one.

Is there a way to set the scaling ?

CAAosPlatform, 2021-04-01:
The aspect ratio between font size and the elements is fixed. There isn't a direct contro for that. However, if you want to try, open circuitSymbols.py and change the following text size related variables:

        # text size and font style
        self.fontSize = 5
        self.fontSizeSmall = 4
        self.textOffset = self.fontSize / 1.5  # offset between symbol and text
        self.textOffsetSmall = self.fontSizeSmall / 2  # offset between symbol and text

fsmMLK, 2021-04-01:
The aspect ratio between font size and the elements is fixed. There isn't a direct contro for that. However, if you want to try, open circuitSymbols.py and change the following text size related variables:

    # text size and font style
    self.fontSize = 5
    self.fontSizeSmall = 4
    self.textOffset = self.fontSize / 1.5  # offset between symbol and text
    self.textOffsetSmall = self.fontSizeSmall / 2  # offset between symbol and text

PinkMushroom, 2021-04-02:
ok. thanks very much. i can play with that.

what about the generation size of the symobls ?

I've been looking through the code and there doesn't appear to be a scale factor.

looks like it uses relative coordinates and hard-coded numbers, e.g.

`inkDraw.line.relCoords(elem, [[-(15.5 + wireExtraSize), 0]], self.add(position, [-9.5, 0]))`

it does return it as a group, maybe some hackery that scales the group automagically when done ?

fsmMLK, 2021-04-02:
yes, the components of the symbols are pretty much hard-coded. Making them all completely parametric would make me go crazy.

If you wish, you can scale elements or groups with the method self.scaleElement()

See the documentation here:
https://fsmmlk.github.io/inkscapeMadeEasy/moduleDefinitions.html?highlight=scale%20element#inkscapeMadeEasy_Base.inkscapeMadeEasy.scaleElement

PinkMushroom, 2021-04-02:
lol. i understand perfectly that you don't want to parametrize everything, i was just making sure i understood the construction.

It seems like scaleElement will do exactly what i want, so I tested it,

`self.scaleElement(group, 0.25, 0.25, None)`

sure enough, it works great.

I was hoping there was a central point that could apply the scaling for everything, but that doesn't appear to be the case. The groups that are being returned by each of the draw routines are being "returned" to inkscape, correct ?
i.e., the scale would have to be added to each routine.

fsmMLK, 2021-04-03:
correct.

The problem with an unified center point is that if you apply a transformation to an element or a  group, then you have to take it into account when defining the center point for scaling that specific element/group.

closed 2021-05-02 by PinkMushroom