fsmMLK/inkscapeCircuitSymbols #26: General does not work but semiconductor works normally

dingyaoshun, 2021-06-04:
The semiconductor works pretty fine. But when I use the General it returns such an error

Traceback (most recent call last):
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\textext\base.py", line 528, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\textext\utility.py", line 253, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command D:\ProgramData\Miklatex\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "circuitSymbols.py", line 567, in <module>
    circuit.run()
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "circuitSymbols.py", line 308, in effect
    self.drawInductor(root_layer, position, value=so.bipoleRLCVal, angleDeg=so.bipoleRLCRot, flagVolt=so.bipoleRLCVolt,
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\circuitSymbols\drawRLC.py", line 256, in drawInductor
    inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint='bc', preambleFile=self.preambleFile)
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1288, in latex
    tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\textext\base.py", line 284, in effect
    self.do_convert(new_text,
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\textext\base.py", line 357, in do_convert
    converter.tex_to_pdf(tex_executable, text, preamble_file)
  File "D:\ProgramData\Inkscape\share\inkscape\extensions\textext\base.py", line 532, in tex_to_pdf
    raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
textext.errors.TexTextConversionError: TeX compilation failed. See stdout output for more details

any help is appreciated

fsmMLK, 2021-06-09:
I could not reproduce this error.

m3nsh, 2021-06-18:
I have the same problem. Everything works fine if I disable latex. I only installed latex to use it for this. Maybe I did something wrong?

m3nsh, 2021-06-18:
I had the same problem. Check this: https://github.com/fsmMLK/inkscapeCircuitSymbols/issues/27

closed 2021-11-16 by fsmMLK

jianghanpeng, 2022-01-15:
I just experienced the same problem. I am running on a Windows x64 platform. It is solved by setting MikTeX's automatic package installation to "Always install missing packages on-the-fly" and then reinstall the TexText.