fsmMLK/inkscapeCircuitSymbols #28: Compatibility with siunitx version 2

Zaunmich, 2021-09-09:
When trying to use inkscapeCircuitSymbols, I run into the following error:

```
Traceback (most recent call last):
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\textext\base.py", line 538, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\textext\utility.py", line 252, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command C:\Users\Bio\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "circuitSymbols.py", line 567, in <module>
    circuit.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "circuitSymbols.py", line 377, in effect
    self.drawSourceV(root_layer, position, value=so.sourceVal, sourceType='general', angleDeg=so.sourceRot, flagVolt=so.sourceVolt,
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\circuitSymbols\drawSources.py", line 79, in drawSourceV
    inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint='bc', preambleFile=self.preambleFile)
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1288, in latex
    tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 140, in run
    self.save_raw(self.effect())
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\textext\base.py", line 286, in effect
    self.do_convert(new_text,
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\textext\base.py", line 362, in do_convert
    converter.tex_to_pdf(tex_executable, text, preamble_file)
  File "C:\Users\Bio\AppData\Roaming\inkscape\extensions\textext\base.py", line 542, in tex_to_pdf
    raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
textext.errors.TexTextConversionError: ! Package keyvalue Error: Unknown option 'per' for package siunitx.
```

The error suggests that a option "per" for the siunitx Tex-Package is provided, which was used in the version 1 of siunitx.
See https://tex.stackexchange.com/questions/327859/package-siunitx-doesnt-work-with-option 

A quick and easy fix is to use version 2 syntax in the preamble definition:
in C:\Users\ *yourUserName* \AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbolsPreamble.tex

```tex
\usepackage{amsmath,amsthm,amsbsy,amsfonts,amssymb}
\usepackage[per-mode=fraction]{siunitx} % change this line here.
\usepackage{steinmetz}
\usepackage[utf8]{inputenc}
%\usepackage[T1]{fontenc}

\newcommand{\micro}{\ensuremath{\mu}}
\newcommand{\phasorName}[1]{ \ensuremath{ \boldsymbol{\hat #1} } }
\newcommand{\impedance}[1]{ \ensuremath{ \boldsymbol{#1} } }
\newcommand{\complexPol}[2]{\ensuremath{#1\phase{#2}}}
\newcommand{\complexPolDeg}[2]{\ensuremath{#1\phase{#2\degree}}}

% commend those lines out, they are the old syntax
% new units
%\newunit{\Vef}{\volt_{ef}} % volt eficaz
%\newunit{\Vrms}{\volt_{rms}} % volt RMS
%\newunit{\Vpp}{\volt_{pp}} % volt peak-to-peak

%\newunit{\Aef}{\ampere_{ef}} % ampere eficaz
%\newunit{\Arms}{\ampere_{rms}} % ampere rms
%\newunit{\App}{\ampere_{pp}} % ampere peak-to-peak

% add those lines, they are the new syntax
\DeclareSIUnit\Vef{\volt_{ef}} % volt eficaz
\DeclareSIUnit\Vrms{\volt_{rms}} % volt RMS
\DeclareSIUnit\Vpp{\volt_{pp}} % volt peak-to-peak

\DeclareSIUnit\Aef{\ampere_{ef}} % ampere eficaz
\DeclareSIUnit\Arms{\ampere_{rms}} % ampere rms
\DeclareSIUnit\App{\ampere_{pp}} % ampere peak-to-peak

% logic Commands
\newcommand{\NOT}[1]{\ensuremath{\overline{\mbox{\ensuremath{#1}}}}}
\newcommand{\AND}{\ensuremath{\cdot}}
\newcommand{\OR}{\ensuremath{+}}
\newcommand{\XOR}{\ensuremath{\oplus}}
\newcommand{\XNOR}{\ensuremath{\odot}}
```

Are there any plans to add siunitx version 2 compatibility?

fsmMLK, 2021-09-12:
Thanks for the message. Yes I am aware of the changes between siunitx 1 and 2.

I updated the repository

Zaunmich, 2021-09-12:
Thanks for the update!

closed 2021-09-12 by Zaunmich