fsmMLK/inkscapeCircuitSymbols #29: Extension greyed out

rahna803, 2021-09-18:
Hi,
I wanted to use your library for my thesis work, I installed all the plugins and extensions, but the option is greyed out as shown.
[image]
any suggestions?

fsmMLK, 2021-09-18:
try opening inkscape from the terminal and look for the output. It might show what the problem might be. Usually it means inkscape cannot find all necessary files to open it. 

which inkscape version are you using?

rahna803, 2021-09-19:
> try opening inkscape from the terminal and look for the output. It might show what the problem might be. Usually it means inkscape cannot find all necessary files to open it.
> 
> which inkscape version are you using?

The error message is get is:

** (org.inkscape.Inkscape:3736): WARNING **: 14:17:36.636: Failed to find resource file 'inkscapeMadeEasy/inkscapeMadeEasy_Base.py'. Looked in:
        C:\Users\ran\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Base.py
        C:\Program Files\Inkscape\share\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Base.py

** (org.inkscape.Inkscape:3736): WARNING **: 14:17:36.659: Failed to find resource file 'inkscapeMadeEasy/inkscapeMadeEasy_Draw.py'. Looked in:
        C:\Users\ran\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Draw.py
        C:\Program Files\Inkscape\share\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Draw.py

** (org.inkscape.Inkscape:3736): WARNING **: 14:17:36.674: Failed to find resource file 'inkscapeMadeEasy/inkscapeMadeEasy_Base.py'. Looked in:
        C:\Users\ran\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Base.py
        C:\Program Files\Inkscape\share\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Base.py

** (org.inkscape.Inkscape:3736): WARNING **: 14:17:36.688: Failed to find resource file 'inkscapeMadeEasy/inkscapeMadeEasy_Draw.py'. Looked in:
        C:\Users\ran\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Draw.py
        C:\Program Files\Inkscape\share\inkscape\extensions\inkscapeMadeEasy/inkscapeMadeEasy_Draw.py

The inkscape version in 1.0

rahna803, 2021-09-19:
well i did end up resolving that issue, now i have a newer issuee:
    import textextLib.textext as textext
ModuleNotFoundError: No module named 'textextLib'

fsmMLK, 2021-09-19:
you have to install the version on 'latest' folder.

rahna803, 2021-11-03:
Installed everything all over again, the extensions work, but they are a bit slow.

closed 2021-11-03 by rahna803