fsmMLK/inkscapeCircuitSymbols #31: Installation suggestion

anthonyjwall, 2022-06-18:
Hi! Thank you for the package(s), it seems really cool and I wish I came across it sooner.

I had an issue on installation that might be worth adding a line into the installation instructions. I was lazy and cloned the git repos into .../inkscape/extensions and then created a symbolic link to the files in .../latest. I also tried copying the files up a level to match your tree shown. 
This didn't work and I was banging my head trying to figure out why - it just seems Inkscape does not like git repos in its extensions folder. I tried this on Ubuntu and MacOS. Works fine when files are just copied from elsewhere. I'd be happy to submit a PR if you'd like.

fsmMLK, 2022-06-20:
I never tried to git clone directly into the extensions folder, I am not aware of any issues with .git folder, but who knows?

However I do create links to my extensions because I do not want to develope them inside .config folder and I don't want to remember to update them every time I modify my extensions.

anthonyjwall, 2022-06-20:
Interesting, I'll try link them from a git repo in a different location... Thanks for the suggestion!:)

closed 2022-06-20 by anthonyjwall