fsmMLK/inkscapeCircuitSymbols #33: It gives me an error, when I clicked Apply

apoorav-singh, 2022-09-19:
Error:

> Traceback (most recent call last):
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py", line 562, in tex_to_pdf
>     exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\utility.py", line 268, in exec_command
>     raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
> textext.errors.TexTextCommandFailed: Command C:\Users\Apoorav\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 646, in <module>
>     circuit.run()
>   File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
>     self.save_raw(self.effect())
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 307, in effect
>     self.drawResistor(root_layer, position, value=so.bipoleRLCVal, angleDeg=so.bipoleRLCRot, flagVolt=so.bipoleRLCVolt,
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\circuitSymbols\drawRLC.py", line 64, in drawResistor
>     inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint='bc', preambleFile=self.preambleFile)
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1310, in latex
>     tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
>   File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
>     self.save_raw(self.effect())
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py", line 287, in effect
>     self.do_convert(new_text,
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py", line 363, in do_convert
>     converter.tex_to_pdf(tex_executable, text, preamble_file)
>   File "C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py", line 566, in tex_to_pdf
>     raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
> textext.errors.TexTextConversionError: TeX compilation failed. See stdout output for more details

I am getting this error on the Inskcape Console:

Configuration;

OS: Windows 10
Python: python 3.10.6
MikTex Console Version 4.8
I am using latest version of "textext" and "inkscapeCircuitSymbols"

What I think is that there may be some missing package in MikTex.

fsmMLK, 2022-09-19:
Yes it seems it is related the miktex. Try disabling latex support and see ifthe extension runs.

apoorav-singh, 2022-09-20:
Yes, the plugin works but I am not able to work out that MikTex problem. If this issue is due to some missing packages in MikTex than what are the packages which are used in this case, and needed to be installed?

It would be really nice if LaTeX can be used.

Thanks.

fsmMLK, 2022-09-20:
you can check the list of packages in the file [circuitSymbolsPreamble.tex](https://github.com/fsmMLK/inkscapeCircuitSymbols/blob/master/latest/circuitSymbolsPreamble.tex), and [basicLatexPackages.tex](https://github.com/fsmMLK/inkscapeMadeEasy/blob/master/latest/basicLatexPackages.tex).

Miktex has an option to allow it to [automaticaly download](https://yihui.org/en/2018/03/miktex-auto-install/) any packages it finds necessary when compiling the text. I'd try this first (the lazy mode =) )

apoorav-singh, 2022-09-20:
Thanks! it worked. I think Lazy mode is way to go! I think beginner like me would have hard time loading these packages manually (or even a user who just needs to draw circuits quickly for their work)

Although it still gives some sort of Warnings:

> C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py:415: DeprecationWarning: inkex.deprecated.main.transform_imul -> Use @= operator instead
>   full_layer_transform *= layer.transform
> C:\Users\Apoorav\AppData\Roaming\inkscape\extensions\textext\base.py:419: DeprecationWarning: inkex.deprecated.main.transform_mul -> Use @ operator instead
>   tt_node.transform = (-full_layer_transform *               # map to view coordinate system

Thanks a lot!
Cheers.

closed 2022-09-20 by apoorav-singh

fsmMLK, 2022-09-20:
This other issue you mention is in textext extension (I do not have any control over it). It is caused by recent changes in python. You can fix yourself this annoyance by openning textext's base.py file and look at line 419 and replace * with @, like. 

tt_node.transform = (-full_layer_transform @               # map to view coordinate system
                                     Transform(translate=view_center) @    # place at view center
                                     Transform(scale=user_scale_factor) @  # scale
                                     Transform(translate=-node_center) @   # place node at origin
                                     tt_node.transform                     # use original node transform
                                     )
                                     
Something similar must be done in line 415.

apoorav-singh, 2022-09-20:
Got it thanks!