fsmMLK/inkscapeCircuitSymbols #34: Can't have a component without value

josejsarmento, 2022-09-21:
OS: Mac OS 12.6
Inkscape version: 1.2.1
inkscapeCircuitSymbols version: [05b434f](https://github.com/fsmMLK/inkscapeCircuitSymbols/commit/05b434fc3f2a1783433662967d24213bdd78a73c)

Hello, I'm trying to place a component without a value, but it keeps automatically adding one. In my case, I want to place a resistor with no value, but it adds a 'Z' by default.

[Here are the screenshots](https://imgur.com/a/Btgz1z0).

I also get a warning whenever I place a resistor, not relevant for this issue but thought you should know.

```
.../org.inkscape.Inkscape/config/inkscape/extensions/textext/base.py:415: DeprecationWarning: inkex.deprecated.main.transform_imul -> Use @= operator instead
  full_layer_transform *= layer.transform
.../org.inkscape.Inkscape/config/inkscape/extensions/textext/base.py:419: DeprecationWarning: inkex.deprecated.main.transform_mul -> Use @ operator instead
  tt_node.transform = (-full_layer_transform *               # map to view coordinate system
```

fsmMLK, 2022-10-19:
I will move this to 'feature requests'.

fsmMLK, 2024-12-25:
done!

closed 2024-12-25 by fsmMLK