fsmMLK/inkscapeCircuitSymbols #35: Error when generating symbols

etnapoli, 2022-10-12:
Installed circuitsymbols on Windows 10.
in some cases it works. e.g. to draw the GND symbol.
In most cases a window opens complaining for an error.

As an exmaple when trying to draw a switch.

"
traceback (most recent call last):
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 562, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\utility.py", line 268, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command C:\Users\etnap\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 562, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\utility.py", line 268, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command C:\Users\etnap\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 646, in <module>
    circuit.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
    self.save_raw(self.effect())
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 477, in effect
    self.drawNPST(root_layer, position, value=so.switchVal, angleDeg=so.switchRot, isPushButton=so.switchPushButton,
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\drawSwitches.py", line 102, in drawNPST
    inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint='bl', preambleFile=self.preambleFile)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1310, in latex
    tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
    self.save_raw(self.effect())
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 287, in effect
    self.do_convert(new_text,
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 363, in do_convert
    converter.tex_to_pdf(tex_executable, text, preamble_file)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 566, in tex_to_pdf
    raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
textext.errors.TexTextConversionError: TeX compilation failed. See stdout output for more details
"

Any hint on what could be wrong?
Thx

etnapoli, 2022-10-12:
This is another example. Itried to generate a '+Vcc' symbol. 
This is the reported error.
=====
Traceback (most recent call last):
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 562, in tex_to_pdf
    exec_command([tex_command, self.tmp('tex')] + self.LATEX_OPTIONS)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\utility.py", line 268, in exec_command
    raise TexTextCommandFailed(message="Command %s failed (code %d)" % (' '.join(cmd), p.returncode),
textext.errors.TexTextCommandFailed: Command C:\Users\etnap\AppData\Local\Programs\MiKTeX\miktex\bin\x64\pdflatex.exe tmp.tex -interaction=nonstopmode -halt-on-error failed (code 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 646, in <module>
    circuit.run()
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
    self.save_raw(self.effect())
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\circuitSymbols.py", line 617, in effect
    self.drawSignal(root_layer, position, angleDeg=so.nodalRot, nodalVal=text, drawLine=so.nodalDrawLine)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\circuitSymbols\drawSignals.py", line 40, in drawSignal
    temp = inkDraw.text.latex(self, group, value, pos_text, fontSize=self.fontSize, refPoint=justif, preambleFile=self.preambleFile)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\inkscapeMadeEasy\inkscapeMadeEasy_Draw.py", line 1310, in latex
    tex.run([r'--text=' + LatexCommands + LaTeXtext, '--scale-factor=1', '--preamble-file=' + preambleFile, tempFilePath],
  File "C:\Program Files\Inkscape\share\inkscape\extensions\inkex\base.py", line 231, in run
    self.save_raw(self.effect())
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 287, in effect
    self.do_convert(new_text,
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 363, in do_convert
    converter.tex_to_pdf(tex_executable, text, preamble_file)
  File "C:\Users\etnap\AppData\Roaming\inkscape\extensions\textext\base.py", line 566, in tex_to_pdf
    raise TexTextConversionError(parsed_log, error.return_code, error.stdout, error.stderr)
textext.errors.TexTextConversionError: TeX compilation failed. See stdout output for more details
=====

fsmMLK, 2022-10-19:
I think you have a problem with your latex install. Try turning off LaTeX support and see if you can generate the symbols.

closed 2024-12-25 by fsmMLK