fsmMLK/inkscapeCircuitSymbols #36: More MOSFET Symbol Options

RicardoSung, 2022-10-23:
Hi, **thanks for your efforts**, it helps me a lot on Mac to draw circuit schematics.
However, I'm an analog IC designer, and **the symbols of MOSFET are a little different from those we usually use in conferences like ISSCC shown below**. So I mean whether you could add some options to add/move the circle of the _MOSFET symbols_.
[image]

fsmMLK, 2022-10-23:
I am not sure exactly what you need. There is already an option to hide the envelope circle around the symbol. Check 'Draw Envelope' option.

You can generate symbols like those in your image by setting 'MOSFET symbol type'  to  '3-terminal (no bulk)'. Please check 'MOSFET symbol type' section in the readme page for detailed description.

I do not know if this this is what you need. Please let me know if I misunderstood you. Can you provide direct examples of what you need and what my plugin provides?

RicardoSung, 2022-10-24:
Oh......I'm sorry. Thanks for your kind help.

closed 2022-10-24 by RicardoSung

fsmMLK, 2022-10-24:
great!
BTW, if you need new symbols you can add a request [here](https://github.com/fsmMLK/inkscapeCircuitSymbols/issues/23)